bambulab/BambuStudio #500: [Feature Request] Exhaust fan control

moebis, 2022-10-23:
When printing ABS, it gives off plenty of VOCs. We have a carbon filter exhaust fan, that doesn't seem to do anything with ABS prints. The ended code should turn the exhaust fan on after the print has completed until the temperature gets down to 35 degrees (at that time everything should have been cleared out of condensed on the surfaces and it's safe to open the printer). This can be done in Bambu Slicer, or the printer controls panels in the slicer and screen for manual control, or make it simple and by default do this behavior at the end of ABS prints. I ended up creating 2 .gcode files that I put on the SD card to do this manually:

fan ON.gcode:

M106 P3 S255
G28 X
G28 Y

fan OFF.gcode:

M106 P3 S0
G28 X
G28 Y

Bpositivedotdk, 2022-10-31:
It would be nice if you could control the fan during the print on the screen.

SaltWei, 2023-02-22:
Always had UI to control the fan. Close

closed 2023-02-22 by SaltWei