bambulab/BambuStudio #501: Slicer error on AMS multi-color part, dependent on part orientation

OlegTheMighty, 2022-10-23:
I posted this in the Discord server and someone suggested that I add it here.
I printed these jeep toys in PLA which worked great, except that 18 of the 20 parts had some blue lines through the gray windows. All 18 parts were rotated the same direction. The other two were rotated 90 degrees and did not have the error.
Painting was done with the bucket tool. I tried repainting the parts and found the same error persisted. Rotating the parts by five degrees appears to have removed the issue.
[image]
[image]
[Jeep Bodies Color with error.zip](https://github.com/bambulab/BambuStudio/files/9847239/Jeep.Bodies.Color.with.error.zip)

OlegTheMighty, 2022-10-31:
Here's another example. similar part, but depending on rotation, the first-layer is divided differently. the bottom is supposed to be blue, and then grey on the front 
[image]

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu