bambulab/BambuStudio #502: [Feature request] Allow me to change Z world coordinate of models

bani6809, 2022-10-24:
I want to stack models, like in this video : https://www.youtube.com/watch?v=RAciAS1ssuc

This is not possible in BS because BS always wants to snap model Z to the bed. Any value you give for Z, BS will overwrite it - it is not possible to position a model in Z space at all.

RobinWhitfield, 2022-10-24:
You can do *exactly* the same as in that video in Bambu Studio.

Right click (on your first part) - Add Part - Load.

randyre, 2022-10-27:
But after you load the part its snaps it to Z0 even if the part should be at a different Z height.  I believe the Z height issue is attached to a different Issue, but I can't find it right now.  Also, the ability to move something in Z is helpful when scaling two parts that need to fit together. Currently to make them line up and test fit I have to export to a CAD program and size them then import back which takes a lot of extra time.

bani6809, 2022-11-03:
basically, don't snap Z for anything unless I specifically request it.