bambulab/BambuStudio #505: [Feature request] allow command line usage (CLI)

Project-J2, 2022-10-24:
We would like to automate gcode generation. Is it possible to implement the command line interface (CLI) function like in PrusaSlicer
([Command Line Usage](https://manual.slic3r.org/advanced/command-line)) ?

lanewei120, 2022-11-02:
currently CLI is still supprted,
however the params have been changed lots. 

we will give a short guide soon and refine them in next release

lanewei120, 2022-12-16:
the CLI guide is published at [BambuStudio's Wiki](https://github.com/bambulab/BambuStudio/wiki/Command-Line-Usage)
BambuStudio v1.4 already support it

Project-J2, 2022-12-17:
Hi,
Thanks for the info, however I can't use it on Windows.
I guess on Windows i need another executable like PrusaSlicer : "prusa-slicer.exe" for the version with HMI and "prusa-slicer-console.exe" for the console version?

lanewei120, 2023-01-11:
> Hi, Thanks for the info, however I can't use it on Windows. I guess on Windows i need another executable like PrusaSlicer : "prusa-slicer.exe" for the version with HMI and "prusa-slicer-console.exe" for the console version?

oops, just found I forgot to reply this
sorry for this.

on windows, you can run bambu-studio.exe also
I have confirmed that it can run on the git bash, 
however, it seems it can not run in windows 'cmd shell'

lanewei120, 2023-01-11:
[image]

closed 2023-03-01 by lanewei120