bambulab/BambuStudio #506: Control Angle Of Support Structures And Interfaces.

wirlybird, 2022-10-24:
It would be great to be able to be able to input the "angle" of support structures and interface areas.  When the interface structure is printed in the same direction as the parts layer above it that can make support removal more difficult.  See picture with yellow arrows
As an example - Simplify 3D does a pretty good job with this. Picture with red arrows.

[image]

[image]

tangketan, 2023-01-04:
Ideally the users need not worry about the line direction of support interface. The algorithm behind should automatically place support interface perpendicular to the upper object layer. As to your case, I'm not sure if it's a bug. Would you please share the 3mf with us?

wirlybird, 2023-01-04:
I'll see what I can do.

Interfaces. (Issue #506)

 

TTVLANCNFSM6AAAAAARM6MLFE> .
<https://github.com/notifications/beacon/AGB3OKHHCA34OWG5NIZS4H3WQTTVLA5CNFS
M6AAAAAARM6MLFGWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTS

wirlybird, 2023-01-04:
Here is the part from my picture.
For this example I used the Bambu default .2mm layer and only change was to
turn supports on so it may look slightly different than the original
picture.

 

<https://github.com/notifications/beacon/AGB3OKHHCA34OWG5NIZS4H3WQTTVLA5CNFS

tangketan, 2023-01-30:
@wirlybird Sorry but I didn't see any model or picture. Would you try again?

wirlybird, 2023-01-30:
I uploaded everything on the ticket system.

tangketan, 2023-02-02:
@wirlybird what ticket number please?