bambulab/BambuStudio #507: [Feature] Exclude Region while printing

Kraminator, 2022-10-24:
This is an amazing feature in Octoprint (using a plugin). When a part of a print gets loose (or a part of a support), this feature lets you put a rectangle or circle put around it, and the printer excludes that region on the build plate. This way, you don’t have to stop the print when any part gets loose and you only loose the part that for loose.

TheHarmory, 2022-10-25:
I was just about to sign up to GitHub to post this as a feature request. 
But I found your post, so now I only signed up to "vote up" :-) 

Really something I would like, and it would be awesome if this could work in both Bambu Studio as Bambu Handy

YifanWuBambu, 2022-10-27:
Skipping the failed objects during printing is in our plan, but maybe not in recent release.

Kraminator, 2022-10-27:
Great, thank you. You are correct, selecting failed objects is even better than defining areas. I am looking forward to this feature, excited that it is on your timeline.

SaltWei, 2023-08-02:
This feature has been included in latest  V01.06.00.00 X1 series firmware and [v01.07.01.62 studio](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.01.62). And please make sure both firmware and studio is updated.

Currently it only work in one filament per plate situation, and doesn't support multi filament because of the complexity of flushing and changing filament. One filament per plate situation can still benefit a lot when has object failed to be printed and need to be skipped.

Thank you every one for this good suggestion!

closed 2023-08-02 by SaltWei