bambulab/BambuStudio #509: Sync Message Acknowledgments

randyre, 2022-10-24:
When a message pops up it currently displays on the LCD, BambuStudio, and Bambu Handy, which is as expected, but when a message is acknowledged or a print is stopped and confirmed we shouldn't have to close all those same messages on other screens or devices. The confirmation should clear all the other same messages.

sodachen22, 2022-10-29:
Thank you for the idea, we will have this feature in the future.

randyre, 2022-12-10:
@sodachen22 Any update on this?

randyre, 2023-01-08:
This is still a major headache for those of us with multiple computers using Studio. I have a bunch of error messages that I've cleared while working at home, and when I get to work and want to work with Bambu Studio I spend the first 5 minutes or more clearing error messages that have been generated by other instances of Studio while away from that computer.

Can we get this assigned please? It's a real headache to deal with on a daily basis.

sodachen22, 2023-01-09:
@randyre We received the requirement, but I am sorry that it does not have priority over some feature development. I think there could be a workaround for your case that you can close studio when you leave.