bambulab/BambuStudio #510: I made this white but slicer says black but preview says white

Benderrodrigu3z, 2022-10-24:
[image]
[Zelda_Keychain.3mf.zip](https://github.com/bambulab/BambuStudio/files/9855075/Zelda_Keychain.3mf.zip)

murrlyn99, 2022-11-01:
I had this issue recently as well. Painted model surface on build plate side of print is correct before and after painting in Bambu studio in prepare tab but after being sliced, some areas of print change color only on build plate face and print that way shown only in preview tab however prepare menu still shows correct painting.

zhimin-zeng-bambulab, 2024-01-18:
The latest version has fixed this problem(e.g. v1.8.4), you can have a try.

closed 2024-01-18 by zhimin-zeng-bambulab