bambulab/BambuStudio #511: [Feature Request] Folder system in My history, Bambu Handy App

Gizmo6, 2022-10-24:
I love the cloud-based My history feature in the handy app.

What would make it even more amazing is, instead of just a pure history in list form to be able to have some sort of folder system for people who frequently print the same set of objects over and over again. This way I could start a print project again at ease without having to manually scroll through the entire history each time.

Allram, 2022-10-25:
+1 on this.
This would help tremendously on multiprinting and it makes it easier to stack often used prints in different folders for easier access.

Superbobo75, 2022-10-27:
But mostly to correct the incomplete print history. Yesterday I printed 6-7 times the same thing and I only see it 1x in the history. In case I am printing on a job so I can't quickly look at the history of what all and how many times I printed. I don't understand the logic behind the Print History.

SaltWei, 2024-01-18:
Move the ticket to right place. Done

closed 2024-01-18 by SaltWei