bambulab/BambuStudio #513: BUG Open 3MF or STL MacOS 10.15.7

Benderrodrigu3z, 2022-10-26:
If I double click the 3mf it doesn't open in any program even though its told to open in BambuStudio, BambuStudio does become the active window but it acts like it doesn't know what to do with the file and just drops it, if I drag and drop the file onto the icon in the dock with BambuStudio open nothing happens, the only way to import is to move the BambuStudio application to one side and drop it onto the build plate but this only works if your on the Prepare Screen...

Benderrodrigu3z, 2022-10-27:
ps this is not a new issue, I just finally got annoyed with it and thought to speak up.

bambu123, 2022-10-31:
Thanks for your feedback. I think it may be the issue with low MacOS version.

Benderrodrigu3z, 2022-10-31:
> Thanks for your feedback. I think it may be the issue with the low MacOS version. We'll check it soon.

not sure how, other programs open properly when you double click the file or drag it to the icon in the dock for that program

bambu123, 2022-11-03:
Double-click the 3mf file when one instance of Bambu Studio opens, and see what happens.

Benderrodrigu3z, 2022-11-03:
Nothing that’s the issue

Sincerely,
Shane F

bambu123, 2022-11-10:
We use a virtual machine to install macOS 10.15.7 but can't reproduce it.
What version of Bambu Studio are you using? make sure to use the latest version.

Benderrodrigu3z, 2022-11-10:
So the newest version made the print menu work better with the vpn on but still small issues with the server on first send. Version 1.3.0.25

Sincerely,
Shane F

bambu123, 2022-11-10:
So that's a new issue. What about the 3mf assosiation problem?

Benderrodrigu3z, 2022-11-10:
Oh sorry I have another ticket open on that issue too, got them flipped in my head

Sincerely,
Shane F

bambu123, 2022-11-16:
@Benderrodrigu3z  So does the problem continue to exist?
Please help send me the app log if it continues.
You can locate the log directory from the menu " Help > Show Configuration Folder" in Bambu Studio.
The log should have these lines when it works.
-------------------------------------------------------
2022-11-16 19:42:53.550596[Thread 0x0000000130a09600]:finished the gui app init
2022-11-16 19:42:53.551684[Thread 0x0000000130a09600]:MacOpenFiles, open files, size 1

bambu123, 2022-11-17:
As you said "even though its told to open in BambuStudio, BambuStudio does become the active window but it acts like it doesn't know what to do with the file and just drops it", Open using BambuStudio force will open BambuStudio, this way there may be some helpful log.

bambu123, 2022-11-17:
Please do not worry. Just try doing some thing with 3mf opening and send us the log.

bambu123, 2023-02-02:
@Benderrodrigu3z 
Does the problem continue to exist? We need your help. Any log is helpful.

Benderrodrigu3z, 2023-03-11:
yes its still an issue, it doesn't even switch to the active screen where the slicer is on open when its defaulted to opening BS program on double click

Benderrodrigu3z, 2023-03-11:
double clicking doesn't do it, dragging it to the dock icon doesn't do anything either, only dragging it to the build plate works

bambu123, 2023-03-13:
Please send us the logs which can be fetched from the configuration folder.
From the menu "Help-> Show Configuration Folder", you can find them.

Benderrodrigu3z, 2023-03-13:
there is no logs on a file not opening, the Mac doesn't even switch screens to the one where bambu is currently open on

Benderrodrigu3z, 2023-03-13:
so when I make 1.5 the default it switches back to the oldest version that I have on file

[image]

Benderrodrigu3z, 2023-03-13:
even with always open with on the right click menu it defaults back to the oldest version
looks like the newer versions don't take presidency

Benderrodrigu3z, 2023-03-13:
if I move 1.3.0.12 to the trash it then defaults to 1.3.0.25 and upon bringing 1.3.0.12 out of the trash that then becomes the default

Benderrodrigu3z, 2023-03-13:
after removing all but 1.3.1.1 and 1.5 the Menu Bar says Bambu Studios on double click but still Doesnt switch the to the active window or open in the app, on removal of 1.3.1.1. it does the same thing to 1.5

bambu123, 2023-03-13:
What's your OS version? The problem may relate to the system because I can't reproduce it here.

BTW:
The default app also stays old on my computer. please remove the old ones when you install the new version, using the 'Replace' not 'Keep Both' option is suggested.
Please do not rename BambuStudio, you can move it to another folder, but do not change its name. The 'New Window' file menu depends on the app name.

Benderrodrigu3z, 2023-03-13:
Scale and Rotate don't listen to the Reset of ZXY
However the Move Command does, see colored markers in pics to understand better

[image]
[image]
[image]

Benderrodrigu3z, 2023-03-13:
I rename it for Version number, things work faster on 1.3.1.1 then they do now
storing it as a zip for now though so it defaults to the new but it still doesn't cause it to switch screens and open even with that override app in System Preferences that I showed above

[image]

bambu123, 2023-03-14:
BambuStudio 1.3.0.12 always opens the new file in the new instance. But it failed to start a new instance because the app is renamed. It seems to do nothing.
BambuStudio 1.5.0.60 opens the STLs in the current window. It is successful.

Is this your problem?

Benderrodrigu3z, 2023-03-14:
I would assume so, is there a way to have the app renamed like you can rename any other app on the Mac and it still would use it?Sincerely,Shane FSent from my iPhone13

bambu123, 2023-03-15:
If that's the rename problem, we can fix it.

Benderrodrigu3z, 2023-03-15:
I haven’t tested thisSincerely,Shane FSent from my iPhone13

bambu123, 2023-04-23:
Fixed in version 1.6, please have a try.

closed 2023-08-23 by SaltWei