bambulab/BambuStudio #514: Feature Request

Benderrodrigu3z, 2022-10-26:
Please add a total time on each color
its a 4 color print, how long is it printing Red for vs Yellow vs White vs Black....
I see filament change times is 109, total time is 5hrs and 23m, how long is it adding to change the filament that many times? Can we also see how much time is spent per color for printing? like yellow is 10min total while black Is 4 hours

YifanWuBambu, 2022-10-27:
Hi @Benderrodrigu3z 
In what senario do you need these detailed information?

Benderrodrigu3z, 2022-10-27:
In every printing scenario.... you can see what color its using more of at a glance so that you can change it to a different color.
The ability to know what is taking the most time so perhaps we can change the color and make it less time total.
If we knew that its 109 times to change filaments well that's say 30sec to change, that's adding a total of about an hour to the print time. If this machine is fast we need to see what is slowing us down from getting models out, perhaps we print half and glue it together instead, if we knew how much time was per color we could then make that inquiry.

YifanWuBambu, 2022-10-28:
@Benderrodrigu3z 
In multi-color printing, the filament change time is ususally the most big part.
We will display the filament change time seperately. Users can adjust color if they think filament change takes too much time.

Benderrodrigu3z, 2022-10-28:
perfect thanks

closed 2022-10-28 by Benderrodrigu3z