bambulab/BambuStudio #516: [Bug] Bambu Studio doesn

whiiiskyy, 2022-10-26:


whiiiskyy, 2022-10-26:
For the matter, I just installed the latest version.

lanewei120, 2022-11-01:
> I added all the common materials to see if I need to change my current profile when moving to X1C but it doesn't display all the options I selected.
> 
> [image]
> 
> [image]

Hello

when the printer preset is selected, only the compatible filaments can be listed in the filament combox
from your picture, 
it is the correct behavior because other filaments are not compatible with the printer "Bambu Lab X1 0.4 nozzle"

if you select the printer of prusa, then the filament with "Prusa..." will be shown then

thanks

H4kunaM4tata, 2022-11-10:
> Hello
>
> when the printer preset is selected, only the compatible filaments can be listed in the filament combox from your picture, it is the correct behavior because other filaments are not [...]

@lanewei120 there is no "compatible filament", any printer will printer Bambu Lab filament as it will print Prusa filament.
What changes is the filament temperature, bed temperature, retraction and etc, that might require some changes.

Removing Prusa filament as option when I select X1C printer just because it is not compatible???
So you are saying X1C printer cannot print Prusa filament!!!

This is such as small detail, I agree but your answer exposed a bigger problem.
What is next, X1C will only accept Bambu Lab filament???

This mentality is 100% wrong!

lanewei120, 2022-11-10:
no, you just misunderstand my point

all these filaments can be used on x1c
however, as there are so many filaments, we only provides limited filament presets which are compatible with the printer by default;(in the filament directory under the vendor's profile)
if you want to use other filaments, you just need to add the filament preset by yourself,

currently each vendor's printer can only be compatible with the filaments under the vendor's filament directory
so if you want to use these prusa filaments on X1C
you can do either of them:
1. copy these filaments into BBL's filament directory, and modify related jsons
or
2. create a user preset based on bambu's compatible filament(named , set the values the same to your prusa filament settings

closed 2022-11-20 by H4kunaM4tata