bambulab/BambuStudio #517: 'Export Sliced File' drop-down menu in the UI is unreachable on Mac

tokamac, 2022-10-26:
BambuStudio 01.03.00.12
macOS Ventura 13.0
MacBook Air M1

The top right button `Print` in BambuStudio, that sends the sliced file to the printer OTA, has also a drop-down menu to choose either `Print` or `Export Sliced File` to write a .gcode file onto an SD card that can then be inserted into the printer.

However on my Mac, this submenu `Export Sliced File` is _unreachable_ because although opening, _it always closes before the mouse cursor can roll over it_. The solution of traditional menus to press and hold the button of the mouse/trackpad does not work because the BS devs have chosen a "MouseUp" event to make the drop-down menu appear (the menu is displayed when the mouse button is released) instead of an "MouseDown" event (the menu is displayed when the mouse button of is pressed).

A workaround currently is to export the sliced file with the File menu in the main menubar handled by the OS (this works)
Please fix this not critical but somewhat annoying UI glitch, which I think has been introduced in BambuStudio 1.3

[image]

Subsidiary question: What is the difference between the G-code file generated by `Export Sliced File…` vs `Export G-code…` in the `File → Export` menu of BambuStudio? Why two G-code files? Is the later a "generic" G-code for other printers, stripped of the X1's specific features?

YifanWuBambu, 2022-11-07:
Subsidiary question: What is the difference between the G-code file generated by Export Sliced File… vs Export G-code… in the File → Export menu of BambuStudio? Why two G-code files? Is the later a "generic" G-code for other printers, stripped of the X1's specific features?

@tokamac 
Export Sliced File - it exports a .gcode.3mf file, which not only contains .gcode file, but also contains the thumbnails, slicing info (especially useful in AMS slot mapping)
Export G-code - just .gcode file, no thumbnails and do not allow to assign AMS slot for each filament in multi-material printing

mvieleers, 2022-12-08:
I have the same issue BambuStudio 01.03.01.01 on MacOS Venture 13.0.1 on MacBook Pro M1 Max

Steps to reproduce:
add an stl to the buildplate
Press "Slice" in top right
The Select the dropdown on the "Print" button
The dropdown opens and shortly shows 2 options:  Print and Export Sliced File. 
As soon as you move the mouse over the dropdown menu it immediately disappears making it impossible to press "Export Sliced File"

[image]

closed 2022-12-16 by YifanWuBambu

Naiki92, 2023-01-10:
v1.4.0.18 - The problem appears again. MacOS Ventura 13.1