bambulab/BambuStudio #519: [Bug] File → Export → Export Sliced File… adds .3mf extension to the .gcode file

tokamac, 2022-10-26:
BambuStudio 01.03.00.12
macOS Ventura 13.0
MacBook Air M1

On a Mac, when choosing `Export Sliced File` (to an SD card for example) instead of sending the print job OTA, BambuStudio automatically adds a .3mf extension behind the correct .gcode extension. The resulting file is considered as a 3MF file, which it is not.

[image]

whiiiskyy, 2022-10-28:
That is actually correct.
Windows here, since for some God knows why reason there is no Linux client ffs, and it is exactly the same.
Once you load that into the printer, it will load the model thumbnail and all.

If you wanna export it as gcode which on this printer adds no value, you have to go to `File → Export → Export G-code`
But when you print it, there is no thumbnail, nothing, it is kind dead looking and it actually overwrote custom g-code I had in place.

So yeah, export the file as it's and you are good to go.

tokamac, 2022-10-28:
You're right. I finally found an SD card and duplicated the same exported sliced file, renaming the first one to file.gcode and the second one to file.3mf, and only the later works properly on the printer with a thumbnail and an estimated print time. So I close this issue which is not an issue!

tokamac, 2022-10-28:
I think however that Bambu Lab should remove the last part of the name ".gcode" inserted automatically before the actual working .3mf extension, which it the thing that confused me in the first place.

tokamac, 2022-10-28:
Last comment which explain this all: 
- The exported sliced file is not a pure .gcode file.
- It's not a pure light .3mf file neither, like the BS project on your hard drive, which weight a few hundreds kB only.
- The exported sliced file is in fact a .3mf file, with a whole .gcode file separately stored inside its metadata structure, that weights several MB! Wow. I learned something.

closed 2022-10-28 by tokamac