bambulab/BambuStudio #521: Application crashes when importing from a SMB disk

fang0654, 2022-10-26:
# Overview

The BambuStudio application throws up a segfault when importing an STL or importing a project from a mounted SMB drive.

# Platform

Windows 10 Professional x64, Bambu version 01.03.00.12

# Details

I am currently using Bambu Studio in a Virtualbox VM, with a mapped drive to my STL storage. When opening the STL, I'm able to navigate to it fine (on the Z: drive), but after opening (and rendering), the application crashes with `Calling AppConfig::save() from a worker thread!`. When restarting, _any_ STL I open will then still crash the slicer.

# Workaround

The workaround I've been using is to open Bambu Labs, save the blank project to a folder on C. This will still crash with the same error message. Then I launch it again, save a blank project to C again, and then I can import an STL from the Z:. Next time I try it will crash again, and I need to repeat the above process. I've attached multiple logs to a support ticket - US221017800001.

lanewei120, 2022-10-29:
hello

I have checked that ticket
it seems the thread name is not the one set in studio, which will cause an exception
and it is related with the virtualbox vm

1. please help to  try the debug version  to confirm this issue( https://www.dropbox.com/s/5vx8fv9krjuyiup/Bambu_Studio_win_public-v01.03.00.21-20221028222821.exe?dl=0),  show the exception picture here
2.  you can also try to build the linux image following https://github.com/bambulab/BambuStudio/issues/12, or use the appiamge provided by others
(I am afraid currently we have limited resources to support linux version, and we could not provide an official appimage in a short time)

fang0654, 2022-11-06:
Sorry for the delayed response, I got the Linux version up and running and have been using it instead. Last night I did fire up the internal build. I got the same results, and the new log files are attached to the same ticket referenced earlier. I hope this helps!

---- On Sat, 29 Oct 2022 06:18:54 -0500 Lane.Wei ***@***.***> wrote ---

hello

I have checked that ticket
 it seems the thread name is not the one set in studio, which will cause an exception
 and it is related with the virtualbox vm
please help to  try the debug version  to confirm this issue( https://www.dropbox.com/s/5vx8fv9krjuyiup/Bambu_Studio_win_public-v01.03.00.21-20221028222821.exe?dl=0),  show the exception picture here

you can also try to build the linux image following https://github.com/bambulab/BambuStudio/issues/12, or use the appiamge provided by others
 (I am afraid currently we have limited resources to support linux version, and we could not provide an official appimage in a short time)

lanewei120, 2022-11-07:
the thread name in the virtual box is "main", it seems  'pthread_setname_np' doesn't take effect in virtual box

we will add a logic to fix this issue in next release

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu

fang0654, 2022-12-18:
Just following up, can confirmed it is all fixed. Thanks!