bambulab/BambuStudio #522: Filament Preset overrides

DocHogan, 2022-10-27:
Feature request: 

Add following setting to Filament Preset->Overrides:

Quality/First Layer Height
Quality/Elephant Foot Compensation
Quality/Bottom Surface Flow Ratio

Speed/First Layer 
Speed/First Layer Infill
Speed/Outer Wall

Additional question...in what file(s) are BL's custom g-codes defined? I'm trying to understand some of what they've done (*especially* G29, which is kinda killing me, but also the several undocumented Gcodes like M1002. TIA)

SaltWei, 2022-10-27:
Thanks for suggestions!
BTW, does other slicer have these same feature? This seems to be odd personally.

DocHogan, 2022-10-28:
No, other "Slic3r" derived slicers don't have these overrides, and I've never understood why not.

These overrides are specifically ones that apply to individual filament printing properties. Print Setting Presets (such as ".16 Optimal @BBL X1C", as included by default) are a good place to hold "one-size-fits-all" settings for these properties, and are good as a starting point as they will generally work with most filaments, though perhaps not optimally.

Every filament type has different printing properties and idiosyncrasies, and even different spools of the same type can be slightly different. PETG likes a slower print and a little more 'squish' on the first layer; elephant foot compensation can compensate for dimensional changes that only occur due to this extra squish. Matte or 'silk' PLA needs a greatly reduced outer wall speed to get the full effect of their visual qualities. This 'Fire Engine Red' PETG from SomeRandomPlasticCompany has slight layher adhesion issues and needs a slightly slower print speed that this "Thomas the Engine Blue" from SomeOtherPlasticCompany. Etc.

Tracking the specifics of each individual filaments optimal settings, remembering to change them in Print Setting Preset, and doing so correctly and not accidentally skipping any, for each and every print...is a lot to do. I guess these adjusted setting could be saved as a new custom Print Settings Preset...one for each and every filament, and again for each and every Print Settings Preset (0.08, 0.16, 0.12, etc)...that's a LOT of custom presets, also incredibly inefficient. But placing these overrides into the Filaments presets is intuitively where they belong and are needed. One preset for each and every specific filament, as needed, and with the overrides needed to print optimally with that filament, all in the right place.

bani6809, 2022-10-28:
ideamaker lets you override just about everything in a preset.
[image]

DocHogan, 2022-10-28:
> ideamaker lets you override just about everything in a preset. )

As does Cura, IceSL, Simplify3d...essentially, *every* slicer but Slic3r-based slicers, as far as I can tell.

aridShrub, 2022-11-18:
I agree right now max volumetric speed is the only speed override at the “filament” level.  It is the same in Pruslicer. I find there are speed overrides at the “nozzle” level, so that if I change from a .4mm nozzle to .8mm nozzle, the “speed” defaults are different.  These defaults are set in the .json files so one might be able to create a dummy “nozzle” for the purpose of being able to change a set of related values.

I also find that slowing down speeds can improve colour print quality, so having different speed defaults for a multi-colour print (vs. A single filament) would also be super helpful.  You could have a model which would use one set of values for layers without a colour change, and automatically use a second set of values if that layer has a colour change! 

I wish I could just tell the printer to slow down right after it does the colour change, (for so many mm of travel perhaps?) because that is the biggest problem with print quality, especially if it is only printing a very short line segment after changing colour.

JPotter73, 2022-11-25:
I agree with first layer flow rate modifier:
https://github.com/bambulab/BambuStudio/issues/548#issuecomment-1327852256

as I said in that post, I'd rather tune 1st layer flow rate for my exact "squish" wants over asking for z-offset adjust (doesn't make sense to have that on a printer that uses the nozzle to probe the bed.)

All first layer overrides can be a good idea for tuning vs artificially lowering z offset from it's measured known value.