bambulab/BambuStudio #523: [SLICER FEATURE] Readable output 3mf file name

whiiiskyy, 2022-10-27:
When exporting the 3mf file, it generates a weird untitled file:

[image]

When exporting it via g-code, you can have the output format like PrusaSlicer:

`{input_filename_base}_{layer_height}mm_{filament_type[0]}_{print_time}_{round(total_weight)}g.gcode`

That outputs the file as `fastbenchy_0.32mm_ASA_30m9s_10g.gcode`
 It is so much better and organized, while looking a folder with dozens of g-code, you can easily identify the file by its material, how long it will take and the amount of filament needed

[image]

Not a folder full of `Untitled_plate_1.gcode.3mf` files.

[image]

Thank you

H4kunaM4tata, 2023-01-10:
SoftFever has implemented this change.
I am closing this.

closed 2023-01-10 by H4kunaM4tata

FoldZero, 2024-04-01:
the setting is missing from latest\current release
[image]