bambulab/BambuStudio #524: Add Print Speed Mode selection to Bambu Studio before sending print job

jjumbc, 2022-10-27:
Right now in order to change speed mode you must start the print, wait for the printer's initial steps, and then select the different speed mode from one of the available interfaces (Printer, Studio, or Handy). 

I think being able to set and forget the mode from Bambu Studio prior to sending the print job would be a useful feature. Thanks!

randyre, 2022-10-27:
Maybe also have a start speed and then a print speed toggle so that you could start a print as normal for say the first two layers and then bump up the speed. Currently you have to sit in front of Studio or the printer and wait for the first layer and then bump up the speed.

Bpositivedotdk, 2022-10-31:
Perhaps also have the possibility to change the speed on the screen to a specific number - between 25-150% would be useful.

zeekens, 2022-11-02:
Yes, so much yes. I've been searching for this setting!