bambulab/BambuStudio #525: BambuStudio should remember printers that have been attached.

randyre, 2022-10-27:
While working remote had a problem connecting to the device. network showed it was still online and Bambu Handy could talk to it. So elected to remove printer and add back but after removal it did not know about the printer and wanted me to use a QR code again but I'm not onsite and have no physical access to printer. It should remember previous devices that have been attached to and offer them as possible devices.

YifanWuBambu, 2022-10-28:
By "removing printer", you mean log out from a printer?
[image]

If so, the printer no longer belongs to that account in the Cloud service and is considered as a new printer. Thus you cannot see it over internet.

Why did you log out your account from the printer? Try to recover the network connection?

YifanWuBambu, 2022-10-30:
@randyre 
Unfortunately, logging out/in does not resolve this problem.
We need to find out the root cause of this problem.

Could you reproduce this problem? If so, Could you help provide the log files.
-------------------------------
How to get log files:
[image]
[image]

randyre, 2022-11-13:
The problem has not reoccurred, but am having lots of WiFi disconnects with the last firmware and software update. Also, Camera/Video is terribly slow now as of .30

closed 2023-08-29 by randyre