bambulab/BambuStudio #528: [Bug] black filament objects impossible to see in prepare

bani6809, 2022-10-28:
it's impossible to see black filament objects in prepare
[image]

lanewei120, 2022-10-28:
Hello

thanks for reporting this issue
1. what is your studio version?
2. Could you give a full picture of your studio with the plate included?
3. could you give the information of your graphics card and the driver version?

RobinWhitfield, 2022-10-28:
@bani6809, this looks more like a graphical glitch displaying the bed surface than an issue with part colour (see #199 for that one).

it should look more like this:
[image]

Benderrodrigu3z, 2022-11-03:
best answer is tell the slicer that's that AMS slot is Gray filament, now you can see and print that slot

closed 2024-01-15 by SaltWei