bambulab/BambuStudio #529: [SECURITY FIX] Why does the camera needs to send data out to work??

whiiiskyy, 2022-10-28:
I'm crazy into security and block any data collection no matter what.

I had blocked two domains the printer was calling to:

- all-d-master-71spctnsfa8p5xvgkssa.iotcplatform.com
- oss-cdn.bambulab.com

The camera stopped working. Bambu Handy can no longer connect with it and after a restart the printer displayed an message saying the camera feed wasn't available.

I had to whitelist what I had blocked, restarted the printer to have the camera back online. That's mad.

Taking into consideration I said no to data collection, no is no.
If I'm using app via wireless to access the camera, the camera MUST work locally instead of some sort of cloud stuff.

This needs to be fixed ASAP.

DocHogan, 2022-10-28:
Agreed! There are needed featured that MUST be supported locally, without WAN requirement. Cam stream, and sending file to printer from slicer are the 2 most critical.

Kraminator, 2022-10-28:
Second that

RobinWhitfield, 2022-10-28:
The camera needs to send data out so that it can be sent to your Bambu Lab account in order to permit viewing it when you're not on the local network. To be clear, this is not anything to do with data collection.

If you switch your printer to LAN mode and log out of the slicer, you'll find that you can still connect to the printer but the video stream doesn't work at all - this is an issue that has been ongoing for a little while but as LAN mode was a late addition it's just something they've not got working yet.
They're also working on a local camera stream accessible outside of Bambu Studio (for which instructions have started to appear on the wiki so it must be close), see #39

whiiiskyy, 2022-10-28:
@RobinWhitfield that is classified as sending data out, full stop.

If the printer needs to call home to provide camera feed using your user account, that is called remote management, meaning, your printer as a network device is exposed to the internet.

Basic users won't have a nice firewall setup or network knowledge so for example, its FTP port could be exploited even so you need the LAN mode pin to authenticate.

I'm not sure if you are into IT but I cannot provide you with a simpler information than above.

Any piece of software MUST ask you permission for remote management.

On that same note, literally everyone is accessing their printer locally via wireless, local network, 5 meters apart, 30 seconds walk, just in case I wasn't clear enough.

Meaning, local traffic MUST always be default, remote management MUST always ask for permission and say "this is a remote management" 

I understand the team is busy with higher priorities, they have been doing a solid work, thank you, but this is wrong and should never need a discussion about it. Full stop.

whiiiskyy, 2022-10-28:
Edit: During my tests, even the timelapse stopped working, blocking those domains generated a 0kb timelapse file which is madness. Or could have been "coincidence".

I'll do more test and set it to LAN mode as long as the timelapse still works of course.

Glad to know tho the team is working on providing the feed via RTSP protocol ♥️🙏

whiiiskyy, 2022-10-29:
Setting the printer to LAN mode, the time-lapse still works as expected.
Only when it is running the default mode, if the domains are blocked, it generated a 0kb file from the time-lapse.

hisptoot, 2022-10-29:
Currently the camera uses the tutk kalay platform. It needs to communicate with tutk's cloud server to do auth, endpoint info exchanging, p2p port mapping and some other staff. 

If both the printer and viewer endpoint are in the same lan network, tutk platform permits the printer to send video stream directly to viewer endpoint. If not in the same lan network, tutk platform will try p2p method first, then try relay-mode. 

That's how tutk sdk works. As far as I know, the packets are encrypted. Maybe there are some exploits. The tutk iotc sdk version is 3.4.4.2.

 If they offer a simple video server in the new firmware, like rtsp/rtmp/hls server, we can get the video stream without cloud staff.

RobinWhitfield, 2022-10-29:
> Any piece of software MUST ask you permission for remote management.

Legally no it doesn't have to do that (anywhere in the world AFAIK), and I can't right now think of any devices that ask you whether or not it's OK for it to connect to remote servers in order to function.

This is no different to basicallly every smart device out there - you connect them to your WiFi and they use their remote servers for all connectivity. There's no local device to device connection going on - everything is routed through remote servers.

I understand your desire for it to not be sending data out by default, but it's not the norm in tech, nor is it expected.

At least Bambu Studio heard everyone's desires for a local mode and have been working on it - they could very easily have said no and just carried on as before. The vast majority of users wouldn't have cared one bit.

whiiiskyy, 2022-10-30:
> This is no different to basicallly every smart device out there - you connect them to your WiFi and they use their remote servers for all connectivity. There's no local device to device connection going on - everything is routed through remote servers.

I don't know which world you are leaving in but in my world that isn't true. As an IT professional, what you said was a big load of s...!

A company router, network switch, smart wireless antenna, wireless camera, you name it, for security reasons remote management is set off by default. 
Never ever through bloody "remote servers" as default.

You have to manually enable, a person with the basic of security knowledge will tell you are 100% wrong!

Thanks to what you believe it's right, there's a website with people's house camera exposed on the internet, people on their sofa being watched by while world without knowing it.

So please, do some security resource to have security knowledge, you have none!

whiiiskyy, 2022-10-30:
Feature already requested: https://github.com/bambulab/BambuStudio/issues/39

closed 2022-10-30 by whiiiskyy

RobinWhitfield, 2022-10-30:
Righto, so how does a Nest thermostat, or a Google Home Mini, or an Echo Dot, or a Hive thermostat, or a Tado thermostat, or a Ring doorbell, etc. work?

All of them are accessed from devices within your home (almost exclusively). All of them push everything through remote servers all of the time. None of them ask you about this - it's simply the default.
None of them (as far as I'm aware) allow you to turn this off, or have a way to access them and use them directly without the company's external servers.

I'm totally aware that this is different to (most - Ubiquiti products as an example are all connected to Unifi by default for remote access) enterprise and (some) prosumer and consumer equipment, but this is not a printer aimed at the enterprise market - it's a consumer product.

Don't confuse devices using remote servers with devices that are connected to the internet and are insecure. As an IT professional, you should understand that these are not automatically the same thing.

whiiiskyy, 2022-10-31:
#39 is streaming via RTSP instead of via cloud which is the problem I reported here so I'm not sure which part you didn't understand.

RobinWhitfield, 2022-10-31:
The local camera stream will be an optional way to view the camera feed, and WLAN mode with remote access will almost certainly remain the default option and the option used by almost all printer owners.

If all you wanted was a LAN mode with local camera access, we could have avoided all of this nonsense discussing devices and remote access. The lack of a local camera feed when using the LAN mode is a problem. The fact there's a remote access mode and the camera gets sent out as part of that is not a problem - it's a feature of the printer.