bambulab/BambuStudio #538: (BUG) Auto Select Last Object and Lay On Face

Benderrodrigu3z, 2022-10-30:
I find myself clicking on the only object on the plate multiple times, I don't print in the center for 2 reasons, reason 1 is that I don't want to warp the plate or always print in the same location, reason 2 is if you put it near the edge you can get it off without bending the plate to make the object detach on Cool Plate

For Example:

BUG 1
Click the object, Move the object, Lay on Face, object moved outside of boundary, Click the object again, Click Move Object again(Need it to auto select last object so we are not clicking the same object multiple times, perhaps a lock object to perspective button?)

BUG 2
One issue here is that when you tell it to "lay on face" the object moves about 100mm on the build plate, why is it not flipping in place? See Photos for Example, the coordinates don't change but the object moves on the build plate somehow? this is a BUG

Example Object
[Bambu Door.zip](https://github.com/bambulab/BambuStudio/files/9896059/Bambu.Door.zip)
[image]
[image]

zengzm1212, 2022-11-16:
@Benderrodrigu3z  
I might think the "BUG 1" is not a  BUG.  The gif below shows what I did as described. Could you describe it more specifically?
[image]

Benderrodrigu3z, 2022-11-16:
The bug is the dimensions arent correct in the slicer so when you flip it it’s flipping what it thinks is the image when half of that is dead non stop space

Sincerely,
Shane F

zengzm1212, 2022-11-16:
@Benderrodrigu3z   Could tell me the steps you generate the 3mf file? 

The cause of the “BUG 2” is the instance center is inconsistent with shape center. The displayed coordinates are the instance center,  and it also flips around the instance center. (But the shape center information is stored in the 3mf file. There is an offset value between the stored shape center and instance center.)

Benderrodrigu3z, 2022-11-16:
I did not make the 3mf, I found it on printables.com

Sincerely,
Shane F