bambulab/BambuStudio #539: Purge Volume auto-calc is confusing

psiberfunk, 2022-10-30:
When I want to alter the default purge volumes, doing so is either tedious (have to change every combination instead of being able to set one value easily), or really confusing.  In particular, the auto-calc button does NOT do what I would have expected:  Taking the values in the chart and scaling by the multiplier... instead it seems to scale by the multiplier AND do something else based on... i'm not sure what logic (color difference or something?).

For example.. here's a before-and-after when I click "auto-calc" when I put 0.75 in the scaling area:

[image]

Changes to:

[image]

The numbers generated make no sense to me, and there's not even a tooltip explaining what auto-calc does.   I expected auto-calc to apply to scaling value uniformly.. not just guess at random color combinations.  Maybe there should be a "scale" button , and a separate "Calc-based-on-colors" button ?

theogott, 2022-11-03:
Yes, and please add a "No Purge" Button, in such case the Filament should not be purged but just continue printing with the other color. As AMS is only Multi Color and not Multi-Material, this should be easy.

randyre, 2022-11-05:
How about if we forget about volume entirely and just set a number of linear mm to purge based on colors. Volume and multiplier and all that fancy engineering stuff is confusing to the average user. I have three multi-extruder machines and in Cura it's just a basic purge _Xmm_ and continue. Having the ability to change the purge based on what colors are used is a great additional feature and should in theory save filament compared to just setting an arbitrarily large number to try and cover all colors but let's keep it simple.

theogott, 2022-11-05:
Generally its a good idea to be able to set anything "in detail" for professional users. 
There should be an easy Alternative way - maybe even just 4 steps like "No Purge, Minimum Purge, Medium Purge, Maximum Purge" that the average user can just choose from.
What these do in detail can then be set "somewhere" for example with the multiplier.

psiberfunk, 2022-11-05:
To be fair the defaults work OK , and we are already talking about an advanced settings panel … so I’m not sure how @randyre ’s logic applies here .   With that being said , the panel is annoying and confusing for two main reasons:

1) Autocalc’s behavior is weird and not explained 
2) there’s no easy way to set a reduced default purge value for all the boxes .. I have to go and alter each box 

I don’t find the concept of purge volume confusing at all. It’s just as easy to mentally digest as linear mm… and for this printer , it’s a fixed ratio because it will only ever use one filament diameter.

randyre, 2022-11-16:
@psiberfunk 

The logic is that every other multi material system and multi extruder printer and slicer that allows purging calculates it in millimeters of filament purged. Now all of a sudden, we have it in volume, which is an entirely different way of calculating the amount of filament purged. Maybe I just need to find a conversion chart and then I'll be able to understand it more. I'm a hardware/network engineer not a theoretical engineer, so my head doesn't do volumetric purging as easily as the concept of linear purging.

YifanWuBambu, 2022-12-03:
> When I want to alter the default purge volumes, doing so is either tedious (have to change every combination instead of being able to set one value easily), or really confusing. In particular, the auto-calc button does NOT do what I would have expected: Taking the values in the chart and scaling by the multiplier... instead it seems to scale by the multiplier AND do something else based on... i'm not sure what logic (color difference or something?).
> 
> For example.. here's a before-and-after when I click "auto-calc" when I put 0.75 in the scaling area:
> 
> [image]
> 
> Changes to:
> 
> [image]
> 
> The numbers generated make no sense to me, and there's not even a tooltip explaining what auto-calc does. I expected auto-calc to apply the scaling value uniformly.. not just guess at random color combinations. Maybe there should be a "scale" button , and a separate "Calc-based-on-colors" button ?

We will optimize the behavior to make it more clear in the next release.

theogott, 2022-12-03:
The wasting of Filament is one of the downsides of the X1.
However i found that if the Filament ends, i can just feed new Filament to the X1 and he will somehow continue to print with it.
I would like optionally the same behavior for the AMS - means zero (or at least near to zero) Filament waste at Change.
Its no problem if we have mixed colors at some place - maybe its even stylish. 
Maybe you can add a "no Filament Waste" Option here which will just "continue to print" with the other Filament. In the same way as if the Filament is "run out".
Let me add that i would also like some Option Checkmarks somewhere, 
to "Continue with other Filament if Filament runs out". 
So i can place some "near to empty" Filament Rols in the AMS and "let it print".
So it will automatically use all the Filament, and if one runs out - take the next.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu

theogott, 2022-12-16:
I have seen that it will automatically contine to print with other filament if the checkmark is set.
Goof feature to get rid of filament rests ...

theogott, 2022-12-16:
I have installed the newest version and it looks still confusing.
Am i doing something wrong?

[image]

I do not see any change.
[image]

YifanWuBambu, 2022-12-16:
@theogott 
The red value are the one that is not in suggested volume range after multiplied with "Multiplier".
The tips below "Multiplier" parameter explains how it work.

The red value are just warnings, you can still print with it if you think it's okay for your case.
[image]

theogott, 2022-12-16:
If i make an Multiplier of "0" does it mean i get (nearly) 0 waste then?
I want to print with as few as possible on Waste.