bambulab/BambuStudio #541: (BUG) Rotate object- Y and Z change values

Benderrodrigu3z, 2022-10-31:
I rotated this object, made is smaller by cutting it then moved it into position, it was then not tall enough so I changed the Z value only to find that the object got larger on the Y access, and when I change the Y access the Z gets Larger, notice the colors for Z and Y have changed, Z is blue, its in the correct locations for moving the object and for rotation but when you switch so the scale menu the values are mismatched from the orientation of the object

[image]

zhimin-zeng-bambulab, 2023-02-09:
We plan to improve coordinate system in the future,  Thanks!

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00