bambulab/BambuStudio #543: Failed uploading print

Bpositivedotdk, 2022-10-31:
I suddenly cannot uplaod my print to the machine. 

It says "Failed uploading print file (code=0, error=)"

My Bambu Handy connects fine....! 
[image]

Benderrodrigu3z, 2022-11-03:
Please see if a VPN is on or possibly the VPN Firewall, I get this when that's on, if not same wifi?

Benderrodrigu3z, 2022-11-03:
Oh perhaps its cause PLA isnt specified to an AMS Slot? if that's it click on the Green PLA Note, see how its a -

StoneLiBambulab, 2022-11-03:
Hi Bpositivedotdk],
    could you upload the log in config folder.
menu-> Help-> Show Configuration Folder。
[image]

upload the log files to debug the issue.
[image]

Stone Li

Bpositivedotdk, 2022-11-03:
Thanks,

I dont have th AMS and I have tried disabling the firewall. 

Also ran the network test in bambu studio and that came out fine. 

The weird thing is I can connect with the machine and see the video feed, from both Studio and Handy. 

Just the file upload and print execution is not working.!

[Log.zip](https://github.com/bambulab/BambuStudio/files/9932285/Log.zip)

Benderrodrigu3z, 2022-11-03:
wanting to rule this out of being a computer issue, Can you try a different computer Or create a new user and trying the print?
I have a VPN that I use from time to time, this is the SAME issues that I get when my VPN is on when hitting print. Its possible that someone could be routing your IP address through them(like a hacker), if a different computer works it might confirm this, it could also be a hosts file error if you were messing with the hosts file at one point that points to 127.0.0.1 as being local

Benderrodrigu3z, 2022-11-03:
Im saying there is a chance, sometimes programs that we have running in the background take our IP and route them through someone else's server, I only have this issue when I upload when my VPN is on, a VPN changes how my IP goes though my laptop out to the world, its possible, not saying you are hacked just saying its possible or possibly a programming running in the background

StoneLiBambulab, 2022-11-04:
Hi Bpositivedotdk,
the log shows upload the url failed and returns 404 code.

http://upload-file.bambulab.com/users/2758071011/models/USdfc1c1bd308d40/profiles/285952/1667465951.3mf?x-oss-traffic-limit=25165824&OSSAccessKeyId=LTAI5tFANCnfNNPNSSLjG97f&Expires=1667552351&Signature=o941g8AahVK9hXrygUek43u4whQ%3D

the info is as the following:
[image]

So the avira blocks the request. Please check your settings of avira to unlock the domian.

Stone Li

Bpositivedotdk, 2022-11-04:
Problem Solved. I added bambu studio to trusted apps in Avira and now it's working :-)

closed 2022-11-07 by YifanWuBambu