bambulab/BambuStudio #544: [Feature Request] - Multiview camera for multiple printers

rhys100, 2022-11-01:
It would be good to see all bambu printer cameras for those who have multiple printers. We are heading to 20 printers and would be nice to have them all on the one screen under device so we can monitor them and click on 1 to expand quickly to full screen

Klober81, 2022-11-13:
This is an excellent feature request and I'd like to provide my perspective from the other end of the spectrum. I'm a home user who only owned one 3D printer (Anycubic i3 Mega S) before I received my X1CC from the KS campaign a bit over a month ago and the X1 hooked me. Since then I've pre-ordered 2 more X1CCs, ordered 3 AMSs and hubs, along with a host of Bambu accessories (plates, filters, etc.) and filament. Basically, I'm trying to convey that your X1 changes the game and I suspect many people - both large and small companies, as well as some individuals - will have more than one X1 and will love having this available to them. It's another thing to set you apart and helps to keep your customers within your ecosystem.

toddanglin, 2022-11-21:
This would be a great feature for managing multiple printers! Today, it takes too many clicks to toggle between printers and check-on print progress.

In the Bambu Studio "Devices" tab, it would also be great to default to an "Overview" dashboard that showed all printers and summary of their current status (ex: Standby/Printing/Error, Time Remaining, etc). Clicking-in to a printer could provide the kind of details we get with the current "Devices" view. I've used Raise3D's "Raise Cloud" for a while and they are a good example of providing UI for efficiently managing multiple printers.

TheCableGuy99, 2022-12-11:
was just about to post this request so +1 for it. The ability to select different split screen views for different numbers of cameras.

niclas1977, 2023-05-08:
This is the number one issue that stops me from buying more P1Ps (I have 7 so far). Bambu should have featured this long time ago. Obviously nothing has happened!

GlennovitS, 2024-04-22:
+1 from me. Any news on this one?