bambulab/BambuStudio #548: Add the ability to reduce volumetric flow rate for the first several layers

misterpeppery, 2022-11-02:
I have a print with steep overhangs that I am printing in ABS and the heat from the bed is causing warping for the first 10 layers or so, but once the print gets far enough above the bed it prints fine. I would like to be able to reduce the flow rate for the first several layers to fix this, but then increase to the normal flow rate so that my print times are not as affected by the slowdown. The best would be if the increase was gradual so that the change in flow rate is not visible in the print. I would like this to be an option in the filament settings so that I can set different values for different filaments more easily.

Benderrodrigu3z, 2022-11-03:
a quick solution is send it then once your happy hit Sport mode, but that is hard on the components and annoying to have to do over and over

Best thing would be for them to add the option:
"Change to Other Layer Speeds at ___th  Layer else stay at first layer speed"

JPotter73, 2022-11-25:
I like this suggestion and the flow rate modifier should (imo):

- allow for +/-% change from default or flow calibrated results
- allow you to specify the number of layers from 1st to apply this to

I find that I don't miss adjusting z offset since this printer probes with the nozzle and that is 100% measured and known value. I'm having great first layer adhesion but some prints I wouldn't mind extra "squish" to close the print lines more. I also see a lot of people complaining about not having access to z offset. This could be a feature to promote instead of adding the z offset in when it really shouldn't be messed with on these printers.

My 2 cents :)

aridShrub, 2022-11-27:
You can change speed or acceleration in different parts of the model, or use shapes as modifiers, giving you control in x,y,z axis; not volumetric but directly speed control. I have used this to implement slower speeds near where colour changes occur. 

I would add a cylinder on the print bed the height would be that of where you want speed settings. Then lower the various line types by 25% for example. Or design the model to have different parts where you need it, not just by z axis.

JPotter73, 2022-11-28:
@aridShrub : Thanks! However, I want to change the flow rate, not speed. Nor the max volumetric.

The intent is to allow people that want more "squish" on the first layer to get it without asking to be able to drop z-offset. Or at least that was my interpretation of the original post.  @misterpeppery can correct me, but I believe flow rate (material setting, percentage modifier) is what was asked to be able to modify for a fixed number of layers.

Note: after writing this, I looked it up and the term I was thinking of is "Flow Ratio"

aridShrub, 2022-11-28:
Flow ratio is like a master speed limit for a filament, it will slow down speeds. It does not change the extrusion it changes the speed. So you can do the same thing with speed control.

If you look at "speeds" after running slicer you can see the effect on the speed graphically. Just match that speed with the speed setting for the line type, wall or infill, etc. 

Doing it with flow ratio just doesn't connect with the intent of that parameter, which is a filament override, not a model override. But I hope you get the setting you want, in spite of its lack of merit.