bambulab/BambuStudio #549: Automated Error resolving for filament retraction issues.

randyre, 2022-11-02:
I've been having a lot of issues with the filament saying it can't or didn't retract.   Usually, it can be resolved by opening the AMS and running filament through manually. Also, once in a while filament is stuck in the extruder/hotend, this has been resolved by removing the PTFE tube and manually inserting filament and then pushing the extrude button on the home screen, then manually push the cutter and pull the filament, then put it all back together and it resumes as normal.

What I would propose is a button or even an automatic option to ignore the error and push filament through anyways for say 200mm, then cut and attempt retract again. This should solve the issue of having to manually fix the error. 

This is especially a problem for remote printing.

Benderrodrigu3z, 2022-11-03:
while this sounds like a good idea the real issue is that the filament has hardened inside the nozzle, if your 10-20 degrees hotter the new filament can push it out or even filament left out, but if this pushes filament through the nozzle at the normal nozzle temp it could break or damage the nozzle and your out a nozzle, for an easier practice print a Y splitter and place it at the rear location, see photo, then all you have to do is feed filament into that and hit either make it cut or with it that hot quickly pull the filament out by hand, this is kind of like a cold pull in that it will pull everything out with it BUT ITS HOT
I recommend the splitter be printed in ABS or ASA as it will need solid threads and at 100% Infill
https://cults3d.com/en/3d-model/tool/bowden-tube-y-splitter

[image]

Benderrodrigu3z, 2022-11-03:
the real issue that your having is that your filament most likely has moisture inside it inside the AMS, there are several solutions to this, the best one being desiccant containers, you print them then fill them with desiccant and stick them in to remove the moisture, also you can place a round Hygrometer in the front so you can see, you want below 20% moisture for a Good Print

randyre, 2022-11-05:
the Y adapter is a great idea, but the issue is that it will fail and stop the printing and unless you have access to the printer (and the Y adapter) there's nothing you can do sometimes for many hours.  An automated system of heating the nozzle up to around 230 and then pushing a few hundred MM through will usually fix the issue without human intervention, and then if that doesn't automatically fix it, then it should pop up the error.

If not an automated system, then at least a button on  the popup asking to retry, cancel, or emergency filament repair

> while this sounds like a good idea the real issue is that the filament has hardened inside the nozzle, if your 10-20 degrees hotter the new filament can push it out or even filament left out, but if [...]

> [image]

randyre, 2022-11-05:
It's brand new filament and I've never had issues like this even with old filament in a standard printer, the pproblem stems from all the manipulation through the AMS pathways is very hard on even the newest and dryest of filaments.  There should be a little more automated logic or at least the option to tgry a few things that might clear the issue without having to have directg access to the printer. Especially on multicolor prints.

> the real issue that your having is that your filament most likely has moisture inside it inside the AMS, there are several solutions to this, the best one being desiccant containers, you print them [...]

YifanWuBambu, 2022-11-07:
@randyre @Benderrodrigu3z 
For firmware issue, please move to Bambu Forum for better support.
https://forum.bambulab.com/

randyre, 2022-11-13:
I posted this over on the forum. Is there also a way that we can force the extruder through the software to extrude a certain amount of filament from a specific slot and have it do so even if the sensors are saying that there is filament loaded that hasn't been retracted properly? 95% of the time, manually pushing filament into the extruder has fixed the problem but had to wait 8-10 hours to get back to the printer to do this. Automation would allow the print to continue as soon as the error is discovered.

AbeFM, 2022-12-04:
Yeah, in manual control, a "go to chute" / extrude XX mm (or extruder absolute position, ideally, so you can advance/retract at will) / wipe, all remote, could solve many problems, without needing all the logic of fully automating a fix.

randyre, 2022-12-10:
I agree, currently when there are errors, all manual controls are locked out and there is not really a way to fix or troubleshoot. That would at least be a way to fix issues, especially if not by the printer. Having it sit there at 140 degrees for 8-12 hours waiting for me to return home seems silly and destructive.

randyre, 2022-12-10:
Also, the ability to tell the AMS to push through filament even though it thinks the path is blocked would push old or broken filament out and could just melt them through the extruder to clear the path. Again, allowing the print to continue remotely without having to wait until you get back to the printer physically.

randyre, 2022-12-26:
@YifanWuBambu  is there a way to see the progress on this since it seems to be moved to the firmware side?

randyre, 2023-07-21:
Is there any update to this?