bambulab/BambuStudio #550: Menu-Window appears on Monitor 0, even if Main Window is on Monitor 1 (PRINT-Button)

theogott, 2022-11-02:
Is a minor issue. If you have Bambu Studio running on Monitor 1 (Right Screen) and you click that 
(Badly designed) Menu left of the PRINT Button to switch the Button to "Save Slicing File" (is here in german).

**Then the Menu-Window will appear on Monitor 0 (Left Screen) even though the Bambu Studio itself is on Monitor 1 (right Screen).**

Possibly there is a "Maximum resolution" builtin that  should prevent the Menu from going outside the Screen resolution, which does NOT include a second Monitor?

**Alternatively i suggest this fix:**
If you are going to repair this, maybe rethink the Button Design.

Currently you need to click 3 times. 
1 Open the Menu
2. choose wether Print or Save
3. Click on PRINT / SAVE to PRINT/SAVE
4. In case of SAVE you will mostly have to set it back to PRINT after this. 
5. And then click on PRINT

In terms of usability these are much too many Clicks. _5 Clicks where 1 Click would be enough._
So i recommend, to Simply make 2 Buttons and avoid the Menu (that tends to appear on the wrongs Screen) at all.
AND you do not even need to make another Button because you already have 2 Buttons (the Menu Button is also a Button).
So resize it, give it the Text "Save Slice" and remove the Menu.

Benderrodrigu3z, 2022-11-03:
forgive me, Im on Mac, but is there not just a slice then print button?
you can save the file or export it to 3mf/stl after your done hitting print if you go back to the prepare window

theogott, 2022-11-03:
It can not be easily explained. I will upload a picture. 
Here are two pictures that show you the issue. See the Menu-Window is on the wrong Monitor.

[image]

[image]

[image]

( Side-Note I use a Multi-Desktop https://youtu.be/a22p7UQ5Qjo 
thats why i could not upload the pictures yesterday, as the Screenshot program will only work on Default desktop, see video.)

CGIwalterwong, 2022-11-24:
Regarding the issue that the menu appears on the secondary screen, can you provide the computer operating system and display parameters? It is better to provide the resolution of the display.

theogott, 2022-11-24:
Anybody who has a Multimonitor System should be able to repeat the problem, just open the Menu on the second screen.
Then you will see that it opens on the first screen.
I think that there is sort of safety-mechanism built in ,that will preven the menu from "leaving the screen"
but using the resolution of the first screen instead of taking note that "Multi monitor systems" are possible.

Also as said a menu is not needed at this place. it can be just 3 buttons, "Slice","Save Slice" and "Print".
This will also make usage easier. 

OS Is Windows 11, both Screens have 4k resolution. As the Menu always appears on the "same wrong place",
this does not play a role. I think the menu just appears at the "Most right place it is allowed wed to appear".
And that is the very right border of Screen 1.

CGIwalterwong, 2022-11-24:
Thank you for your feedback. We will solve this issue as soon as possible.

theogott, 2022-11-24:
If you have any questions, contact me at any time, I am also on Skype.

CGIwalterwong, 2022-11-26:
> If you have any questions, contact me at any time, I am also on Skype.

This will be fixed in the next version.

theogott, 2022-11-27:
I can confirm that with the newest version from today, the Issue ist still there, see:

[image]

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu