bambulab/BambuStudio #551: Step Files crash BambuStudio on import

Benderrodrigu3z, 2022-11-02:
[FL_X1_top lift.STEP.zip](https://github.com/bambulab/BambuStudio/files/9922648/FL_X1_top.lift.STEP.zip)

SaltWei, 2022-11-03:
Can't reproduce this issue on the latest release version. My importing operation is shown as gif.
[image]

SaltWei, 2022-11-03:
What version or platform do you use? Mac or windows platform. My PC platform is windows.

Benderrodrigu3z, 2022-11-03:
macOS Catalina 10.15.7
MacBook Pro 2012
16gb 1600 mhz ddr3
12tb ssd storage
2.6 ghz quad core intel core i7

Benderrodrigu3z, 2022-11-03:
Please try the Zip file, the author has since fixed it on the site and removed the step files

SaltWei, 2022-11-10:
Thank you very much for sharing!

Can't reproduce this issue on the mac computer I find.
[image]
Is it possible to update the system of you mac? Since it's really hard for me to find a 2012 MacBook.

SaltWei, 2022-11-11:
OK. Thnaks for response! I will try to find a 2012 to reproduce again.

Benderrodrigu3z, 2022-11-11:
I have the same crash, I had to have someone else convert the file for me however fusion 360 can also convert it

Sincerely,
Shane F

SaltWei, 2022-11-25:
Fixed patch has been merged. Please wait for the next release version.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu