bambulab/BambuStudio #552: Different Type SLICE Filament BUG

Benderrodrigu3z, 2022-11-03:
In Order to Print from AMS slot 4 in ASA slot ONE Must be ASA as well, it CAN'T be PLA even if Im not using it, even if there is no filament in it, it doesn't care was in 2 or 3 but 1 must be TOLD its the same type on Slice

[image]
[image]

SaltWei, 2022-11-03:
A known issue when add negative part. Will be fixed in next release

Benderrodrigu3z, 2022-11-03:
sounds good thanks, closing this

closed 2022-11-03 by Benderrodrigu3z