bambulab/BambuStudio #553: Slicing Problem?

theogott, 2022-11-03:
I have a 7 MB File where the slicing seems not to work properly. 
Who can i send it to?

Benderrodrigu3z, 2022-11-03:
upload a photo or gif of the issue here, if its an stl step or 3mf zip the file then drag it onto the Write screen, when you hit comment it will show it, if you want to see how it looks the preview screen will show you

theogott, 2022-11-03:
[Uploading Slicing-Problem.zip…]()
Please take a look, it.
I can choose normal or hybrid support but the result is not satisfying.

YifanWuBambu, 2022-11-07:
@theogott 
Seems it is not a solid model.
[image]

theogott, 2022-11-07:
Its directly out o the 3D Scanner. But .. it should slice somehow.
Whatever it is, it should slice and print ... not?

YifanWuBambu, 2022-11-07:
The generated mesh is not a solid mesh, which is required by the slicer.

theogott, 2022-11-07:
Ok, then the Slicer should make it solid or help he user to understand what he is doing wrong. 
How can it be made "solid"?
Is it a big deal? 
Can the slice offer me "Shall i make this to a solid mesh"?
And I click "Yes please"?
Most users probably do not know the difference, and this is just "Out of the scanner".
So Scanner -> 3D-Printer = 3D Copier
I believe that should work somehow.

YifanWuBambu, 2022-11-07:
We can do some research of how to support this kind of mesh, but it may not be solved in recent release.

For now, it is suggested to check whether the scanner vendor has provided such a tool.

theogott, 2022-11-07:
Ok, so a closed Mesh is currently needed to be able to slice things.
On long run anything should
a) be possible to slice
or
b) I need a clear error message that tell me what to do.
Maybe on the short run that is possible?
Does the slicer Know what the problem is?
Can he tell me "This mesh needs to be filled" or something?

SaltWei, 2024-05-06:
> Maybe on the short run that is possible?

Hard to do this, should be handled from the source of mesh.

closed 2024-05-06 by SaltWei