bambulab/BambuStudio #558: [BUG] Printer warnings/errors not displayed correctly in BS

cdsmakestuff, 2022-11-03:
[image]
Bambu Studio fails to show actual error/warning message in Studio popup.

Benderrodrigu3z, 2022-11-03:
there is no info here as to what you did before hand, hard to know what happened

candre23, 2022-11-05:
> there is no info here as to what you did before hand, hard to know what happened

That's the problem - this happens whether you "do anything" or not.   I've had similar (maybe the same?) weird errors pop up at random times during a print.  Sometimes the printer stops printing, sometimes it doesn't.  It should never be throwing an error with no context, but that's exactly what it's doing.

StoneLiBambulab, 2022-11-11:
Hi @cdsmakestuff  and @candre23,

Could you help to upload the studio log for me. here is the guide to get the studio log

https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

Thanks!

Stone Li

candre23, 2022-11-11:
Here's all the recent logs.

[log.zip](https://github.com/bambulab/BambuStudio/files/9986000/log.zip)

StoneLiBambulab, 2022-11-21:
Hi @candre23 ,
    The language code is not set correctly. I will fix it in next release. You can try to switch to another language and switch back to your language. 

Thanks!

Stone Li

debren27, 2022-12-08:
I've had this same problem for a while. Finally decided to look into it. Thanks for the easy fix!

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu