bambulab/BambuStudio #559: [BUG SLICER] Slicer keeps deleting user preset

whiiiskyy, 2022-11-04:
From time to time when opening the slicer, the user preset is gone:

[image]

[image]

[image]

Fix: Manually copy the files over from default to my user profile

[image]

[image]

Restart the slicer:

[image]

[image]

This cannot be right?!?

Benderrodrigu3z, 2022-11-04:
I feel that however there is a Save button to save the current settings, its to the right of that drop down in the last photo, see the Floppy Drive, that's there save button next to the Magnifying glass.
If you have the printer on and you told it what filament and color is in which slot when you hit the dropdown for filament it will display those options but they won't be filled in so you have to hit + 4 times then select them each individually, I get your pain its the same on the MacOS version. Just trying to help, perhaps your info with this will help them fix this bug.

H4kunaM4tata, 2022-11-04:
> I feel that however there is a Save button to save the current settings, its to the right of that drop down in the last photo, see the Floppy Drive, that's there save button next to the Magnifying glass.

I actually noticed after opening this ticket that my preset is incomplete. There are profiles missing, changes that I had to hunt down screenshots to set them again.

Atm, after every change I do, I copy the user folder as backup.

For what it is worth, I do not have cloud sync enabled, I don't like cloud stuff and local management is a must, but it seems like it get lost somewhere and just deletes your profile.

Somebody mentioned that the fact I have a folder called 11xxxx means my slicer is logged which is true. 
On that same note, logging out from the profile will use the default profile and in theory no more issues but that also means you cannot access the camera anymore and because the camera isn't a true network camera, you cannot access it via RTSP with VLC for example.

lanewei120, 2022-11-09:
Hello @H4kunaM4tata 

thanks for reporting this issue

could you upload the user presets to us for a check?
also please help to upload your logs when these preset are deleted

thanks

H4kunaM4tata, 2022-11-10:
@lanewei120 could you please let me know where to find the log??

This is not a printer issue, it is a slicer one and I couldn't find a log option.

At the moment I am not logged but slicer still saves the presets under a folder with my Bambu Lab account id. 
If I delete it, it should use the default one but instead it recreates my user folder without the presets.
I have no cloud sync enable so that couldn't be the issue either.

I believe that is the problem.

Thank you

lanewei120, 2022-11-10:
the user presets folder of me:
[image]

the log folder of me:
C:\Users\lane.wei\AppData\Roaming\BambuStudio\log

there are some .info files under your presets folder
it includes some state of syncing
so I need these files besides the logs to have a check

thanks

lanewei120, 2022-11-18:
@H4kunaM4tata hello
are these issues still there?
could you help to upload the logs and presets for us to have a debug?

H4kunaM4tata, 2022-11-18:
> @H4kunaM4tata hello
> are these issues still there?
> could you help to upload the logs and presets for us to have a debug?

@lanewei120 sorry for the lack of reply. I'll backup my presets first before logging back in and testing this again ASAP.

H4kunaM4tata, 2022-11-20:
@lanewei120 
The problem seems to be the slicer getting lost when cloud syncing is disabled.

[image]

This is how my profile looks like when I am logged. 
It creates a folder with an ID and any change is saved there.

[image]

When I logout, my user folder is deleted, the default folder is then set as default

Once I log back in:

- it creates my user folder back, an empty folder since I have no cloud sync enabled, that is expected
- it uses whatever presets the default profile has
- any new change is them saved in my user folder from now

My user presets is gone, once I log back in.

[image]

If you logout, any preset change gets deleted.

The slicer should not delete the user folder preset once you have logged out or at least ask if you wanna keep it or delete it.
It just deletes the folder on its own.

To avoid this behavior, do not login:

[image]

Restore the backup I started doing to the default profile: ( I started doing a backup every time I change anything to avoid getting my presets deleted )

[image]

All presets are back:

[image]

For security reasons, the logs were edited.
I am not sure how I feel about log keeping both ID and token in plain text.
I am not logging in again, all the logs were deleted.
No application should ever log those piece of information, it is a security issues.

[log.zip](https://github.com/bambulab/BambuStudio/files/10048935/log.zip)

StoneLiBambulab, 2022-11-23:
Hi @H4kunaM4tata ,
 i will fix it in next release and remove the token from logging file.

Thanks!

H4kunaM4tata, 2022-11-23:
> Hi @H4kunaM4tata ,
>  i will fix it in next release and remove the token from logging file.
> 
> Thanks!
> 
> 

Thank you for that, I'm closing this ticket.

closed 2022-11-23 by H4kunaM4tata

YifanWuBambu, 2022-12-16:
This is included in v1.4.