bambulab/BambuStudio #560: Jagged edges and missing layers in 10.3

kebabskal, 2022-11-04:
[image]

Edges have random jaggy shifts and sometimes appears as if a layer is missing.
It looks like some layers are randomly shifted in the z-direction.
It seems to happen with any of the standard settings.

**Steps to reproduce:**

1. Add primitive cube
2. Rotate cube 10 degrees around the X axis
3. Click Slice
4. Observe the error

SaltWei, 2022-11-10:
Thank you very much for reporting issue to us!

Could you share a 3MF file for us to reproduce?

kebabskal, 2022-11-10:
Tried reinstalling the same version that this used to happen on, but can't seem to replicate anymore. Very strange.
It was consistently happening when I posted the issue.

Closing since I can't replicate.

Thanks!

closed 2022-11-10 by kebabskal