bambulab/BambuStudio #561: Inconsistent modifier behavior

lvnr, 2022-11-04:
I need an ability to print a part with open top and one side, while retaining the infill, so that I can pour a liquid from two sides into the printed part.

Adding a modifier on top of the object and setting top layers to 0 works great as seen in the screenshots below:

[image]
[image]

Either the topmost layer or the layer where the modifier meets the object is not closed with a top shell, as I expect. (as you can see, the infill is visible all the way through the part)

But as soon as I do the same with a modifier on the side of the object, and set wall layers to 0, the result is not the same as with top layer modification, because it closes the wall where the modifier and part meet.

[image]
[image]

Is there any way or workaround to have wall layer modification to behave the same as top layer modification?

Benderrodrigu3z, 2022-11-04:
question, how did you create the square to start this? did you import an stl or add part Cube?
Trying to help.

YifanWuBambu, 2022-11-07:
Hi @Benderrodrigu3z ,

You can apply it like this. Why the behavior of wall is different with top surface? The wall extrusions need to be a closed loop for better printing effects.

[Slicing-Problem.zip](https://github.com/bambulab/BambuStudio/files/9948054/Slicing-Problem.zip)
[image]
[image]

Benderrodrigu3z, 2022-11-07:
> Hi @Benderrodrigu3z ,
Tag the original poster posting the issue, not someone else that's trying to help

Benderrodrigu3z, 2022-11-07:
@YifanWuBambu, can you try Tagging the original poster posting the issue, not someone else that's trying to help...

YifanWuBambu, 2022-11-07:
> @YifanWuBambu, can you try Tagging the original poster posting the issue, not someone else that's trying to help...

Ooops. Thanks for the reminder.

lvnr, 2022-11-09:
@YifanWuBambu thanks for your suggestion, although it's helpful for a simple model like a cube, i'm not sure i could apply the same to a complex shapes like I have.

I'm attaching below one of the models I need to print.
[Example model](https://github.com/bambulab/BambuStudio/files/9973874/Back.4.4.zip)

In order to achieve what you did with the cube, I would need a similarly complex model that would fit inside this model, which will be quite difficult to produce for all my files.

[image]

[image]

@Benderrodrigu3z In the example above I used the built-in Cube model, for demonstration purposes, so it's not imported.