bambulab/BambuStudio #562: (Bug) Project Overwrite Protection -Delete Not Reset Project Name

Benderrodrigu3z, 2022-11-04:
I removed everything from the build plate, when I added a new item I then saved it not seeing that I saved it OVER the last Project and lost all edits. If I clear the build plate the named save file should go with the delete all or Delete if its the last object.

[image]

Benderrodrigu3z, 2022-11-09:
still a bug on the new version, perhaps a do you want to start a new project window popup if you remove the last object along with an undo button to bring that back

kvnper, 2022-12-10:
Understandable confusion as Cura saves as new file every time you save (there is no dedicated just "save" button, only "save project". So moving from Cura to bambu studio is a learning experience for Cura users for the different behaviour. I would personally like to see it done the same as it is done in Cura - just my personal preference.