bambulab/BambuStudio #563: UI - Control buttons not responding when clicked

marqueone, 2022-11-04:
When creating a new project, in the prepare viewport none of the control buttons that are enabled function. Unable to add a model using the add model button.

When creating a new project, after adding an STEP or an STL model to the build plate none of the control buttons at the top of the prepare viewport, or the control buttons on the build plate function.

At present this is reproducible 100% of the time and requires me to switch to a windows VM where the same problem exists, though with varying success.

[image]

OS: macos monertay v12.6
BambuStudio: v1.3
hardware: intel MacBookPro16,1

YifanWuBambu, 2022-11-07:
Hi @marqueone ,
You mean after clicking the tools on the top bar, none of them show submenu or any effect on the 3D objects like this?
[image]

I did not reproduce this issue on v1.3.0.12.

marqueone, 2022-11-07:
@YifanWuBambu that's right. both the icons across the top of the view port, and the icons on the side of the build plate. Today I've not run into this issue but all last week I couldn't get any of the buttons to work. I'm not certain if its working now because I just did a clean install of macOS Ventura.

YifanWuBambu, 2022-11-07:
Okay. I will record this bug internally. If you encounter it again, please create a new bug
 here.

SaltWei, 2023-10-08:
If you encounter it again, please create a new bug. Thank you very much!

closed 2023-10-08 by SaltWei