bambulab/BambuStudio #564: [Feature Request] Add option to always require second factor authentication when logging into Bambu accounts

BambuLabDiscord, 2022-11-04:
I would like to be able to always require second factor authentication whenever logging in in Bambu Studio since we use it in a public makerspace.

Benderrodrigu3z, 2022-11-05:
just pick a better password nub

Benderrodrigu3z, 2022-11-05:
adding an option to make everyone else annoyed is hardly worth there time, just pick a stronger longer password

Benderrodrigu3z, 2022-11-05:
Perhaps just add a button that resets all loggins which would log everyone out of your account

hifihedgehog, 2022-11-05:
> adding an option to make everyone else annoyed is hardly worth there time, just pick a stronger longer password

So what I asked for is to allow users on an individual basis within their account to toggle 2FA. That's why I say it is an option and not a strict requirement.

guillaumemolter, 2023-02-07:
+1 for optional MFA. This is an important piece of security for a machine that can potentially burned down your house.