bambulab/BambuStudio #565: [Issue] Slicing an octagonal ring crashes Bambu Studio

BambuLabDiscord, 2022-11-05:
Slicing the file "Stackable_Core_Lock_Ring.stl" from the following free project causes a crash: https://thangs.com/designer/3DPrinty/3d-model/Rolling-Storage-Box-31111

The model in question is nothing but a very small octagonal ring.  It only crashes if laid flat - if I stand it up on end then it can be sliced perfectly fine.  I think it has something to do with the Brim, because setting Brim Type to "No-brim" also prevents the crash.  I am running version 01.03.00.12

Scarfmeister, 2022-11-07:
What size brim are you using? I am running the same version version of Bambu Studio as you and it sliced fine with Auto, Manual, and No-Brim. I am using a 5mm brim and a 0.1mm brim-object gap.

However, when trying to slice with a 0.8mm nozzle, it gave me an error "One object has empty initial layer and can't be printed." It also gave me a warning before hand "The model has too many empty layers." It may be a model issue at certain layer heights and wall thicknesses.

mctribble, 2022-11-08:
I’m still on mostly default settings.  Brim type auto, 5mm brim, 0.1 object gap, 0.4 nozzle… so I went back to see if I could reproduce it, and now I can’t anymore.  Maybe it was some other stability issue on my system or a combination with other stuff I had on the plate, but I can’t remember now exactly what else was going on at the time.  Are there log files somewhere that might say something useful?

lanewei120, 2022-11-09:
hello

thank you for reporting this issue.

we have also tried this stl with lots of params, and can not reproduce this issue.
there should be a crash_xxx.log in the log folder,
however, as there is no pdb files in the release version, we can not find useful information from there

if there is a 3mf to reproduce it, will be great helpful

thanks

mctribble, 2022-11-09:
Sadly i do not have that either.  I would say go ahead and close this as unreproducible and if I see it again I'll make sure to note down everything.  I really wouldn't be surprised if it was on my end anyway.  My machine has a memory leak from some sort of driver (no clue which) and it had been a few days since last reboot when the issue occurred.

Thanks,
Christopher "Kaleb" Hake

SaltWei, 2023-08-16:
Thanks! Please reopen if it can reproduce again.

closed 2023-08-16 by SaltWei