bambulab/BambuStudio #567: Acceleration limits are being ignored for the extruder.

RobinWhitfield, 2022-11-05:
I've been testing with some foaming PLA, and to get the best results out of it the acceleration of the extrusion of material into the hot-end has to be done very slowly.

As a test, I set `M201 E1` in the filament start G-code. The slicer correctly informs me that my 9 minute model will now take over an hour as the extruder very very slowly gets up to speed, but if I send this model to print, the acceleration value is totally ignored and the print finishes in ~9 minutes.

I even tried changing the machine limit for the extruder to 1 as well with no change.

This issue occurs whether I send a print directly from Bambu Studio, or export the G-code and load that onto the machine via SD card.

SaltWei, 2022-11-23:
This is confirmed to be firmware bug, rather than studio issue.
Please wait for the update of firmware.

RobinWhitfield, 2023-01-18:
@SaltWei, any updates on when this bug may be fixed?