bambulab/BambuStudio #568: Possibility to change filament in the layer without AMS - M600 support

Superbobo75, 2022-11-05:
Currently it is not possible to simply change the filament without AMS if I only want multilayer printing with different colours. M600 support would be sufficient or if I specify multiple colors and when I click on the plus next to a layer in the sliced model and add a filament change to the layer, it is ignored without AMS and the model is printed with one color to the end. The Marlin M600 code is ignored completely. It would be enough if a pause is added, which is the only one that works, to be able to do a filament unload at least for that pause - unfortunately the option is out of order - in the case still without AMS.

RobinWhitfield, 2022-11-05:
I'd also love for M600 to be supported. The mechanism for unloading filament is already in place, so that can be done automatically (and if the filament cutter poker has been removed it can just error and ask the user to do it manually).

There's been a real lack of user guided anything on this printer so far and it's a shame. Not everything has to be automatic, especially when it can just end up being quite opaque and very difficult to troubleshoot if something goes wrong.

Once #160 is added, it'll make it nice and easy to print on a raft of dissimilar plastic (which right now I've been doing with a different slicer)

bolsoncerrado, 2022-11-08:
+1 for M600!

Sebazzz, 2022-11-08:
Especially useful if you want to do filament change with filaments unsupported in the AMS, like TPU!

Wob76, 2022-12-20:
+1 for M600

SchrodingersLad, 2023-01-08:
+1 for M600

Gr4sil, 2023-03-03:
+1 for M600
Also there should be possibility to put M600 middle of the layer => you could make multicolor/multimaterial prints without AMS by changing the filaments manually

bolsoncerrado, 2023-03-03:
Yep that's the whole point!

ThatOldOne, 2024-11-03:
+1 for manual filament change during print. I'd also like to see it allow a switch to different material.