bambulab/BambuStudio #570: [FIRMWARE BUG] Firmware overwrites slicer changes

H4kunaM4tata, 2022-11-06:
Take PETG as an example.

This is the Generic PETG profile 
[image]

I am using FTP to sent the files to the printer, everything works fine and no issues

[image]

But I wanna use my profile,

[image]

 I increased the vitrification temp and when I copy to the microSD card and try to print, the printer shows a vitrification warning.

If I send the same file to the printer via FTP, it shows a 0kb and I cannot print.
The only change was the vitrification temp so clearly the printer is overwriting the slicer settings or there is something else behind all of this.

The printer must accept the slicer values, and not the other way around.

Ironically, my custom PLA profile works just fine but anything else I send via FTP, the printer sees as 0KB

[image]

I have to remove the microSD card, copy the files from my PC to it, and then the printer accepts it.

`vitrificatio_ASA.3mf` is the very same file I sent over via FTP so something is happening, some sort of firmware verification or I don't know and that is setting the files invalid.

[image]

H4kunaM4tata, 2022-11-06:
Okay so the problem could be lack of refreshing the printer cache or whatever it is called.

I had deleted all the models above via FTP and send another test in ASA.
I went to the printer and it was still showing the models I had deleted previously and the ASA was even being displayed.

So I removed the microSD card, connected it back and the files were gone as expected and the printer recognized the ASA model.

For what it is worth it, I am using a SanDisk Extreme V30 class, it is a microSD designed for 4K/8k recording and the same one I use everywhere, GoPro, Phone, Tablet, Raspberry Pi, you name it, so microSD speed isn't the issue

YifanWuBambu, 2022-11-07:
Hi @H4kunaM4tata 
For firmware bug, please move to https://forum.bambulab.com/ for better support. Thanks!

Superbobo75, 2022-11-08:
> I am using FTP to sent the files to the printer, everything works fine and no issues

How does one send a file via FTP? Without pulling out the card and inserting it into my computer, I have no way to save anything to the card.

H4kunaM4tata, 2022-11-09:
> > I am using FTP to sent the files to the printer, everything works fine and no issues
> 
> How does one send a file via FTP? Without pulling out the card and inserting it into my computer, I have no way to save anything to the card.

Set the printer to LAN mode under the network tab, it displays a pin aka password.

FTP to your printer IP address, use that pin and the username bblp.

Make sure your microSD is attached to it, FTP only gives you access to the microSD card root.

Voila. Upload/delete 3MF files to the printer locally without cloud bs, download timelapse to your pc.

Make sure you have a solid microSD to avoid slow speed aka files getting corrupted or printing issues, I've SanDisk Extreme V30 class for 8K video recording. It's 128GB and I do not remove the microSD anymore, no bloody cloud stuff either. 

There's a catch, LAN mode disable Handy and Slicer, meaning you cannot manage the printer from them.
Personally, I don't care and I don't like devices going out to the internet without reason instead of using local network.
My MK3S+ didn't require babysitting neither should my X1C.

As for the camera feed, while BL don't sort this out, I'll have my network camera via RTSP with VLC.

Allram, 2022-11-09:
> Set the printer to LAN mode under the network tab, it displays a pin aka password.
>
> FTP to your printer IP address, use that pin and the username bblp.
>
> Make sure your microSD is attached to it, FTP [...]

You can actually disable LAN mode again and rebind the printer to the cloud and still have FTP access :⁠-⁠)

psiberfunk, 2022-11-17:
Technically I think FTP transfers are an undocumented “feature” that’s not officially supported ?

H4kunaM4tata, 2022-11-17:
> Technically I think FTP transfers are an undocumented “feature” that’s not officially supported ? 
> 
> It’s certainly useful , but it’s not really valid to file a bug against something that isn’t a supported feature ..

Vitrification is a slicer feature so I'm not sure where your pov is coming from.

closed 2023-01-10 by H4kunaM4tata