bambulab/BambuStudio #571: Allow re-assembly of multi part models

randyre, 2022-11-06:
When adding multipart models it asks to combine them into one model, but then you lose the multi-part multi-color aspects and then have to paint to color.  

Conversely, you can tell it no and keep it as a multipart, but then you have to assemble the parts like a kit after printing and some parts may not even be printable standalone.

Solution 1. would be to allow all the parts to remain separate, allow you to assign an AMS slot to the individual parts and then allow assembly

Solution 2. Do as Cura does and even though the parts are assembled in the correct placement, they remain separate and are allowed to be assigned to different extruders (or in this case AMS slots)

RobinWhitfield, 2022-11-06:
If you split a model and move parts around, you can select everything and right click - "Assemble" to turn parts back into a multi-part model.

However I'm not sure why you would split the multi-part model in the first place, when you can move each part separately from object view.

Or am I fundamentally misunderstanding what you're trying to do?

YifanWuBambu, 2022-11-07:
> If I accidentally move a part on a multipart model it's really hard to get it all aligned again. Please provide a way to re-assemble the objects once it's been split into parts and moved, just like when you first import the files.

Does undo (ctrl+z) help?

randyre, 2022-11-09:
The problem is when you click on assemble it reassembles it into a solid object at the location they have been moved to. I would like it to reassemble it back just as if I had imported it initially as a multi-part model.

randyre, 2022-11-09:
> If you split a model and move parts around, you can select everything and right click - "Assemble" to turn parts back into a multi-part model.
>
> However I'm not sure why you would split the [...]

I'm talking about both imported multipart models, as well as taking a single solid model and having the slicer turn it into multiple objects or parts.  This facilitates coloring the object way faster than painting it, but once it's broken into parts if you move any part, it won't reassemble, It will just turn it back to a solid with the parts in their last position.

As for musing control-Z if I take a single object and move it out of the group to say paint it I can no longer use Control-Z as it will not only move it back but also undo the paint.

Currently, other slicers can reassemble multi-part models after they've been imported. Why limit re-assembly only to initial import?

randyre, 2022-11-09:
[deadpool-multicolour-remix-for-mmu-and-palette-model_files.zip](https://github.com/bambulab/BambuStudio/files/9972514/deadpool-multicolour-remix-for-mmu-and-palette-model_files.zip)

Import these files, then after it's imported move one of them out of the group to maybe paint it on a spot. Then try and get it reassembled for printing. If you try and hit center it drops everything to the build plate. you assemble it turns it back into a solid object.

YifanWuBambu, 2022-11-10:
> [deadpool-multicolour-remix-for-mmu-and-palette-model_files.zip](https://github.com/bambulab/BambuStudio/files/9972514/deadpool-multicolour-remix-for-mmu-and-palette-model_files.zip)
>
> Import these [...]

@randyre For this model, you do not need to move one of them out of group for painting.
Just specify the filament (color) for each part of the whole object in object list is enough.
[image]

YifanWuBambu, 2022-11-10:
If you want to choose each part more easily, please enter the Assembly View.
[image]

Here you can easily select each part of an object by simple click and paint(specify color) by numeric key.
It is also support to explode the object to make painting more easy.
[image]
[image]

randyre, 2022-11-10:
I'll try that. If it does what I hope it does I'll close this ticket.

randyre, 2022-11-10:
OK, close, but not 100% usable.  Need to be able to explode the object farther apart to make it useful. Also, you can only change the colors of each object, there is no paint allowed, so if say within a black object I want to change the color of the say the deapool logo on his chest I can only change all of the object not just a portion of it.
If we could explode an item in the assembly view, then go back to prepare to paint it, then back to the assembly view to re-assemble that would suffice.

randyre, 2022-11-10:
Also, once the files are imported there's no way to assemble them properly unless I chose that in the initial import.

randyre, 2022-11-23:
Still having this issue. Using center does not work either as the parts are different sizes. Would really like to have the ability to re-assemble parts after importing them or moving them.

YifanWuBambu, 2022-11-23:
> 

Do you mean this?
[image]

YifanWuBambu, 2022-11-23:
If it is still not what you want, could you help record a video for how other slicer handle this?

randyre, 2022-11-25:
Yes, but select all but one object. Allow it to import and make one model with multi parts.  Now go back and pick the one part you didn't import and try and make it fit and line up where it's supposed to go. All it will do is import and drop it onto the build plate.

If you still need a video, I need a recommendation for software to record the screen with other than using my phone.

randyre, 2022-11-25:
https://user-images.githubusercontent.com/60475095/203905738-33d8d419-0b7f-41fe-8944-99108b1a7df7.mp4

YifanWuBambu, 2022-11-26:
@randyre 
Thanks for you explanation. I got what you mean. It has been added to our plan list.

randyre, 2023-01-16:
@YifanWuBambu 

Any chance this is going to be assigned to someone and make it in the next build?

YifanWuBambu, 2023-01-19:
> @YifanWuBambu
> 
> Any chance this is going to be assigned to someone and make it in the next build?

Hi @randyre ,
I added it to next month's schedule, but not in high priority.

randyre, 2023-07-21:
@YifanWuBambu  June, and have had a couple of releases since the last post. Any chance this will be looked at for an upcoming release?

bolsoncerrado, 2023-07-21:
We're finishing July already actually lol

Topless-Stang, 2026-09-29:
Has it been added in 2026 yet?

bolsoncerrado, 2026-09-29:
WHoa! Time flies! LOL