bambulab/BambuStudio #572: AMS Lettering A1 B1 C1 D1

Benderrodrigu3z, 2022-11-06:
Some of us are going to be getting the 4 set of AMS's
With so many it will be difficult to sometimes remember which one is slot 1 vs slot 5 vs slot 9 vs slot 13.

The real concern here though is its easier to know A B C D where its A1 A2 A3 A4, B1 B2 B3 B4, C1 and so on for the Slicer.

That way we can go that ams is A that one is B that one is C.... and if WE want to add a label we can, don't force that on us though. Perhaps an option at checkout though for those that want it and a photo so we can see where the emblem goes, only so se can know if its going to cover up the hydrometer that we put in it.

Benderrodrigu3z, 2022-11-08:
I just ordered 3 AMS, to have this is an option in the preferences menu so I can just add a label of ABCD or but them in abcd order would be great then counting all the way down to find which bay is slot 10 for instance

YifanWuBambu, 2022-12-03:
This will be in the next release.

Benderrodrigu3z, 2022-12-03:
great thanks

YifanWuBambu, 2022-12-16:
This is included in v1.4.