bambulab/BambuStudio #574: Cancel Objects option

Azariah98, 2022-11-06:
I’ve got a very long print running with multiple parts on the print bed. Some of them have fallen over and spaghetti’d, but some are just fine. I used to have a plug-in for octoprint on my Enders that allowed me to cancel some objects and keep the others.  This would be an incredibly useful feature to add to the X1.

Benderrodrigu3z, 2022-11-06:
I myself asked for this feature awhile back, they said that it would be coming soon

YifanWuBambu, 2022-11-07:
@Azariah98 
This is in our plan list, but may not be in the recent release.

JPotter73, 2022-11-16:
upvote on this please :)

TheCableGuy99, 2022-11-17:
Experienced this myself so is a definite extra vote from me! Please implement it asap.

SaltWei, 2023-08-02:
This feature has been included in latest V01.06.00.00 X1 series firmware and [v01.07.01.62 studio](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.01.62). And please make sure both firmware and studio is updated.

Currently it only work in one filament per plate situation, and doesn't support multi filament because of the complexity of flushing and changing filament. One filament per plate situation can still benefit a lot when has some objects failed to be printed and need to be skipped.

Thank you every one for this good suggestion!

closed 2023-08-02 by SaltWei