bambulab/BambuStudio #576: Making the Camera IP viewable

Benderrodrigu3z, 2022-11-06:
Is there a way to access the ip/camera address so that we can set a tv or Roku on it so we can always be watching it.

For security please add a 4 digit passcode option to the device itself for Camera Viewing from IP address.
Best location would be under the Network tab as we need that to get the IP address for the camera.
Perhaps adding it as an on screen boot up question with the next firmware update so people know its now an option.

Example:
10.0.1.35/camera

YifanWuBambu, 2022-11-07:
Hi @Benderrodrigu3z ,
This is a firmware feature. Please move to https://github.com/bambulab/BambuStudio/issues/576 for better support. Thanks!

ACJUK, 2022-11-08:
+1 for this!

Benderrodrigu3z, 2022-11-08:
> +1 for this!
https://forum.bambulab.com/t/making-the-camera-ip-viewable/904
Feel free to comment on it, there is also a Heart button on there once you create your username

closed 2022-11-08 by Benderrodrigu3z