bambulab/BambuStudio #577: Replace error codes with words

Scarfmeister, 2022-11-07:
If there is an error code, I should be able to tell what the error is from the slicer. It's great that the dialog box pops up, but I have no idea what the issue is until I go to the printer. It would be great if there was a hyperlink to the Wiki page for that error like when you scan the QR code on the printer, even if the page currently isn't completed.

We should also be able to press continue from the dialog box if it's something like spaghetti detection and was a false positive (basically anything that wouldn't require user interaction at the printer).

YifanWuBambu, 2022-11-07:
@Scarfmeister 
Could you take a screenshot of the code that does not go along with a text explanation?
This may be a bug.

Scarfmeister, 2022-11-12:
[image]
[image]
[image]
[image]
[image]

Sorry it took so long, had to get a print job that would produce error codes.

YifanWuBambu, 2022-11-12:
Thanks a lot! We will fix it in the next release.

StoneLiBambulab, 2022-11-14:
Hi Scarfmeister,
    Could you please help to upload the studio log. thanks!

Export studio log as following:
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

Stone Li

Scarfmeister, 2022-11-17:
[Log_Files.zip](https://github.com/bambulab/BambuStudio/files/10032815/Log_Files.zip)
Here's log files from the 4th to this morning.

Scarfmeister, 2022-11-21:
[BambuStudio.zip](https://github.com/bambulab/BambuStudio/files/10052109/BambuStudio.zip)

StoneLiBambulab, 2022-11-21:
Hi @Scarfmeister，

    The language code is not set correctly. I will fix it in next release. You can try to switch to another language and switch back to your language.

Thanks!

Stone Li

closed 2023-10-08 by StoneLiBambulab