bambulab/BambuStudio #580: Keep top surfaces out of sparse infill volumes

Your-Saviour, 2022-11-07:
I am not sure whether this is a feature or a bug. If it is a feature it would be nice to have a toggle as it can affect part quality, this will make more sense later.

**Problem**
This all started when I was trying to print this part
[image]

As you can see, it has a raised bezel and a large recessed area.

Upon taking  it out of the printer I noticed this distant line
[image]

I've done my best to take a photo of it but it's more pronounced in person (Can provide more photos), it looks exactly like a slight layer shift and that's originally what I thought it is.

**Cause**
During slicing Bambu slicer is putting a top surface throughout the entire recess including under where the bezel needs to go.
[image]
See how the top surface is being printed directly underneath the infill for the bezel. This is creating a noticeable line in the final product.

This is a recreation just using simple shapes and a negative part.
[image]
The top surface is printed larger than what it needs to be.  This could be a setting that I have missed but I have asked around the discord already and no one can find anything. 

This is what I think it should look like
[image]
This is sliced in Cura using the standard Ender 3 0.16 profile. As you can see there is no yellow top layer printed underneath the bezel.

Can provide a .3MF file upon request it just won't let me upload it. However, it should be easy to reproduce.

Talking a guess this is done for part strength, It would be nice to have the option to set how much extra top surface is printed as this does affect print quantity on certain parts.

SaltWei, 2022-11-07:
Could you share the model for us to reproduce?
Thank you for reporting issue for improvement!

Your-Saviour, 2022-11-07:
Here you go! :)
[this-is-broken.zip](https://github.com/bambulab/BambuStudio/files/9948645/this-is-broken.zip)

Your-Saviour, 2022-11-07:
Sorry didn't mean to close this

Your-Saviour, 2022-11-07:
It doesnt seem to let me reopen this, I don't know why the default is to close it if I comment lmao

SaltWei, 2022-11-08:
> The top surface is printed larger than what it needs to be

Yes. This is same with prusa, and we guess the purpose is avoiding the situation that internal bridge has no anchor.
Cura indeed has better line handling which can make internal bridge be anchored well, without enlarge the area too much. And we will learn from it.

But anyway, this is not the root cause of this issue. The rootcause is 25% infill/wall overlap. When print fastly, the two walls are not cooled enough, and be squeezed when print infill, expecially when print solid infill.
1 Reducing infill/wall overlap, but this may cause reverse pulling and gap between infill and wall.
2 Printed slowly. I reduced inner wall, outer wall and internal solid speed speed to be 100 mm/s.
Printed with my X1C. The final model looks better now.

[image]

Your-Saviour, 2022-11-08:
Thank you for your time looking at this issue, 
I hope something good eventually comes from it.

closed 2022-11-08 by Your-Saviour

SaltWei, 2022-11-16:
Cura is without the expansion of top surface, but also has other side effect. See the floating small solid infill inside sparse infill.
[image]
Prusa also has same with issue. But with the expansion, it's higher possibility to be printed successfully.
This is a model shown below.
[image]
Interesting issue and will keep on it.

Your-Saviour, 2022-11-16:
Thanks for continuing to look at this, appreciate you and the team's effort! :)