bambulab/BambuStudio #581: [Bug] first layer travel acceleration should not apply to non-printing moves

bani6809, 2022-11-07:
First layer acceleration setting is applied to all commands, including non-printing moves.

This makes non-print moves on first layer incredibly slow. There seems to be little reason to do this.

SaltWei, 2022-11-08:
This is not a bug. And it meets our expectations.
Too high travel acceleration may make the extrusion line of first layer be pull-off.

bani6809, 2022-11-09:
how about making this an option so I can decide for myself if i want fast non-printing travel moves?

after all, brim is also an option.

SaltWei, 2022-11-25:
Is any printing quality issue or problem caused by this?

SaltWei, 2023-10-07:
> how about making this an option so I can decide for myself if i want fast non-printing travel moves?

Will add it into settings in developer mode